Gnarus-G/maccel #9: Mouse cursor doesn't move on the Y axis

anthrazitrr, 2024-03-22:
After i installed maccel and rebooted, the mouse cursor stops moving on the Y axis.

https://streamable.com/n8x69e

OS: Arch Linux 6.8.1-zen1-1-zen
Server: Xorg with libinput

Gnarus-G, 2024-03-25:
Wow that is wild. I have same setup on my daily driver - same os and server you listed. So, I don't think it's that.
But I'll test it out in a VM.

I guess I should have some kind optional tracing setup for situations like this. I can't infer enough from the video.

@anthrazitrr Which mouse are you using? Can you try another one.

Gnarus-G, 2024-03-25:
Wild. What was your suspicion about opentabletdriver?

So, I'll have to set up a debug build for this. If you are willing, you can use it to troubleshoot.

anthrazitrr, 2024-03-25:
> Wild. What was your suspicion about opentabletdriver?
When i was using leetmouse it was breaking the tablet driver.

> So, I'll have to set up a debug build for this. If you are willing, you can use it to troubleshoot.
👍🏻

Gnarus-G, 2024-03-28:
> After i installed maccel and rebooted

Btw, rereading your first comment, does that mean it was fine before you rebooted? - or you didn't use your mouse until you rebooted.

As for debugging, so far here's what we can do:

Can you clone this repro, and checkout the debug-build branch: `git checkout feat/debug-build`
Then, at the root of the repo, you
- `sudo make uninstall` if you had it installed, probably not
- `sudo make debug_install`
- `sudo dmesg -w` and move your mouse around, 
   - taking note of the line move input log lines, like "[MOUSE_MOVE] input              (x, y)"

What I expect is that as you move your mouse 
- left, x is negative
- right, x is positive
- up, y is negative
- down, y is positive

0 means no movement. The faster you move it the bigger the absolute value of x/y. 

What do you notice when you try it? I would appreciate a video like last time. Let me know if I need to clarify these steps further.

anthrazitrr, 2024-04-02:
https://youtu.be/ilfrL0JSkVQ

Sorry for not replying for 5 days, here's the video.
Every time i move my mouse to the left, kernel log reports y as -1, which is very strange.

Gnarus-G, 2024-04-02:
> Sorry for not replying for 5 days, here's the video.
Absolutely no worries at all.

Thanks for the video. I have a suspicion that I'll look into.

anthrazitrr, 2024-04-07:
Also, could the problem be caused by Glorious mice relying on a separate driver in the kernel?

https://www.phoronix.com/news/Linus-Not-So-Glorious-Mouse

Gnarus-G, 2024-04-09:
Yeah, after reading that now I think the normal hid driver in linux does something for this mouse that I am not doing. Btw my suspicion from before didn't really pan out.

I am gonna buy that mouse and use it for debugging. If you could hook me up with an direct link, like an amazon link or something, that'd be cool.

I think I can spend some time on it this weekend.

anthrazitrr, 2024-04-10:
I've bought it in a local store in Ukraine, so unfortunately I can't help you with that.

anthrazitrr, 2024-04-19:
I have tested every mouse I have:

Don't work at all with maccel:
Logitech M650 Wireless
Bloody T7 Winner

Misbehaving input:
Glorious Model O/O-
Hator Stellar Pro Wireless (reports as Telink Stellar PRO Mouse)

Gnarus-G, 2024-04-20:
Yean I am sorry to hear that. 
Now that I have played around with the Glorious Model 0- I bought, I found out that the problem is that some mice report the movement values in very different formats. 

Leetmouse was kind of handling it with a custom parser for different types;
So it will be an adventure for me to port over linux's handling of all different mice.

ParanoIIa91, 2024-04-21:
Hey, i really got excited when someone posted your project on discord, because mouse accel is essential for me, but when i activate it my vertical movement gets locked and it becomes scrolling wheel instead lol, i am using Lamzu Thorn 4k.

Gnarus-G, 2024-04-22:
Alright guys, @anthrazitrr, @ParanoIIa91

I have ported over leetmouse's parser over in another branch.
If you can, try it and let me know if it helped you.

Basically:
- clone this repro
- checkout another branch: `git checkout feat/leetmouse-hid-parser`
- `sudo make uninstall`: if you had it installed, probably not
- `sudo make install`
- `sudo maccel bindall`: in case it maybe didn't bind automatically.

ParanoIIa91, 2024-04-22:
I am not very good with Linux yet, but this is what i am getting?
[image]

Gnarus-G, 2024-04-22:
@ParanoIIa91 No that's my mistake: wrong command.

Use `git clone https://github.com/Gnarus-G/maccel.git`

ParanoIIa91, 2024-04-24:
[image]

anthrazitrr, 2024-04-24:
> ![Screenshot_20240424_093901](https://private-user-images.githubusercontent.com/74189594/325125987-aeed2443-81ac-450d-b1c7-bc1566a4e0b6.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTM5NDg5MTksIm5iZiI6MTcxMzk0ODYxOSwicGF0aCI6Ii83NDE4OTU5NC8zMjUxMjU5ODctYWVlZDI0NDMtODFhYy00NTBkLWIxYzctYmMxNTY2YTRlMGI2LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA0MjQlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNDI0VDA4NTAxOVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTkzNTZjODEyMWIzMzMxZjM3NmM0MjdiYzkzZGZmNTI5YTFjNTQ1MWVmMjgwYWFkMzVmNjIwMWJlY2IwMTdhNzMmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.d7s0GSjTt9F_8hla97Khx8RmQmS3OKoms6vA7TTpWGs) mouse cursor still behaves weirdly but i dont think those commands did anything, maybe i am doing something wrong?

@ParanoIIa91 
Delete the maccel folder that you already have:
`rm -rf maccel`

Then clone the repo again:
'git clone https://github.com/Gnarus-G/maccel.git'

Change the current directory to the newly cloned maccel folder:
'cd maccel'

Checkout the branch that has the experimental HID parser:
`git checkout feat/leetmouse-hid-parser`

Uninstall the previous version:
'sudo make uninstall'

Install the new version:
`sudo make install`

The new HID parser works for me, I have tested Hator Stellar Pro Wireless and Glorious Model O- so far.

ParanoIIa91, 2024-04-24:
Thank you very much, its actually working! First time in my linux experience, i found a proper mouse accel project, that actually works, now i just need to mess with settings, because i am not sure yet, how exactly doest it work :)

Gnarus-G, 2024-04-24:
I am glad to hear it's working for you both. I will keep this issue open for now and direct other people to it when applicable. Depending on how long it takes to research a better fix, I might merge that branch and put this parser behind a flag.

Gnarus-G, 2024-04-24:
@ParanoIIa91 
> how exactly does it work

This issue is not the place go down this line of inquiry, but...
Depending on what your specific confusion is, checkout the [Readme](https://github.com/Gnarus-G/maccel?tab=readme-ov-file)
Especially the [References](https://github.com/Gnarus-G/maccel?tab=readme-ov-file#references)

Gnarus-G, 2024-06-24:
Noting here that you can install `maccel` with the leetmouse hid parser, simply with:
```sh
curl -fsSL https://www.maccel.org/install.sh | sudo MACCEL_LEETMOUSE_HID_PARSER=1 sh
```

ParanoIIa91, 2024-07-03:
> > ![Screenshot_20240424_093901](https://private-user-images.githubusercontent.com/74189594/325125987-aeed2443-81ac-450d-b1c7-bc1566a4e0b6.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTM5NDg5MTksIm5iZiI6MTcxMzk0ODYxOSwicGF0aCI6Ii83NDE4OTU5NC8zMjUxMjU5ODctYWVlZDI0NDMtODFhYy00NTBkLWIxYzctYmMxNTY2YTRlMGI2LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA0MjQlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNDI0VDA4NTAxOVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTkzNTZjODEyMWIzMzMxZjM3NmM0MjdiYzkzZGZmNTI5YTFjNTQ1MWVmMjgwYWFkMzVmNjIwMWJlY2IwMTdhNzMmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.d7s0GSjTt9F_8hla97Khx8RmQmS3OKoms6vA7TTpWGs) mouse cursor still behaves weirdly but i dont think those commands did anything, maybe i am doing something wrong?
> 
> @ParanoIIa91 Delete the maccel folder that you already have: `rm -rf maccel`
> 
> Then clone the repo again: `git clone https://github.com/Gnarus-G/maccel.git`
> 
> Change the current directory to the newly cloned maccel folder: `cd maccel`
> 
> Checkout the branch that has the experimental HID parser: `git checkout feat/leetmouse-hid-parser`
> 
> Uninstall the previous version: `sudo make uninstall`
> 
> Install the new version: `sudo make install`
> 
> Pro tip: if something doesn't work, look at the output of the command.
> 
> The new HID parser works for me, I have tested Hator Stellar Pro Wireless and Glorious Model O- so far.

[image]
This method also doesnt work for me?

ParanoIIa91, 2024-07-03:
> @ParanoIIa91 Delete the maccel folder that you already have: `rm -rf maccel`
>
> Then clone the repo again: `git clone https://github.com/Gnarus-G/maccel.git`
>
> Change the current directory to the newly [...]

[image]
Doesnt work anymore? I am on Opensuse Tumbleweed.

ParanoIIa91, 2024-07-03:
> Noting here that you can install `maccel` with the leetmouse hid parser, simply with:
> 
> ```shell
> curl -fsSL https://www.maccel.org/install.sh | sudo MACCEL_LEETMOUSE_HID_PARSER=1 sh
> ```

This method unfortunately also doesnt work for me?
[image]

Gnarus-G, 2024-07-03:
@ParanoIIa91  Like the output says, you have `make` installed. I understand you're new to Linux, so get comfortable command-line outputs. You're gonna want to grab make from your distros' package manager.

ParanoIIa91, 2024-07-03:
> @ParanoIIa91 Like the output says, you don't have `make` installed. I understand you're new to Linux, so get comfortable with command-line outputs. You're gonna want to grab `make` from your distros' package manager.

[image]
I installed "make" but this time i am getting different error?

Gnarus-G, 2024-07-03:
@ParanoIIa91 You are having an issue that is wholly unrelated to the main issue that this thread is about. Your distro is probably one that I'm not familiar with, seems very barebones. Refer to this section of the Readme, https://github.com/Gnarus-G/maccel?tab=readme-ov-file#linux-headers, and install the appropriate Linux headers for your kernel version.

If you still need help, I can help you more directly on discord: https://github.com/Gnarus-G/maccel?tab=readme-ov-file#linux-headers

AlexanderVanDamme, 2024-07-21:
with my model d wireless (also from glorious), i only can move my mouse up and down

[image]

i tried using the install with the leetmouse parser but it didn't seem to do anything (yesterday i tried to make leetmouse work but there my mouse didn't move at all after rebooting my pc)

any idea? i fear model d might send different format's than the model o

pretty hyped about this project, if i get it working that would be awesome, was even thinking about implementing raw accel's motivity (sigmoid function acceleration curve) if you're not planning to do that yourself

Gnarus-G, 2024-07-22:
@AlexanderVanDamme Are you saying the leetmouse driver works fine for you, while maccel with the leetmouse parser doesn't?

No idea right now. Maybe you're right.

AlexanderVanDamme, 2024-07-22:
No, I tried leetmouse and there my mouse didn't work at all. Not sure why.

When trying maccel, my mouse only goes up and down. I didn't notice any difference between having the leetmouse parser installed. But I might have to try again.

I'm quite new to linux. When should I restart my pc, to make sure the changes have any effect? For example, should I restart after reinstalling maccel with the different parser, having first installed default maccel?

Gnarus-G, 2024-08-05:
@AlexanderVanDamme I would appreciate it if you tried #23 and see if your mouse works with that version of the driver. It's a re-implementation and it seems promising, but I need help testing it so far.

AlexanderVanDamme, 2024-08-05:
I tried it and it works like a charm! Thanks so much!

Gnarus-G, 2024-08-05:
@AlexanderVanDamme No way, really?! I'm glad! Let me know if you feel any weirdness (eg. input lag). I'll continue polishing it up and once I'm ready I'll come with new install process and release it.

To your earlier question about developing features, I'm open to it for sure. I'll only ask that you write up or talk me through any changes first because if I don't understand it then I won't ship it. You can reach directly in the discord server https://discord.gg/5ZAb4ESY

AlexanderVanDamme, 2024-08-05:
I joined the discord. For now I haven't yet tested it in game, I just installed that branch, changed the params and I saw they had effect. I don't feel any input lag, I will notify you if i feel anything weird.

Gnarus-G, 2024-08-10:
I just released the new driver. Just reinstall maccel, and you should be good to go.

closed 2024-08-10 by Gnarus-G