Gnarus-G/maccel #12: Values change after closing cli interface

NeonNAX, 2024-04-25:
Hate to be that guy but I have an issue:

When I fill in my values they all look as they schould I then press q as instructed.
When I then reoppen the cli, some values a different.
 
Steps to reproduce
-Input:
--SensM 0.0917
--Accel 0.002048
--Offset 78.125
--OC 2.02399
-quit out of cli
-reopen cli
-see some new values:
--SensM 0.08984375
--Accel 0.00390625
--Offset 78.125
--OC 78.125

OS = arcoLinux (arch but with pasases)
Kernel = 6.8.7-zen
WM = Hyprland
Terminalemulator = kitty

My numbers are this low bc I use 10000dpi.
Relating to my dpi would it be posible to have a way to rescale the graph so higher then 1600 dpi would be easyer to use?

p.s. I am dyslexic, hope its readable and helpful

Gnarus-G, 2024-04-25:
It's normal to lose some precision in the numbers due to not using real floating point numbers. I couldn't get floating points to work in the kernel.

That being said, the change in your accel and output cap are very significant. So I'll see what I can do to get more precision. Let me see a screenshot of your graph.

NeonNAX, 2024-04-25:
Here you go:
WhatI fill in:
[image]
What it ends up as:
[image]
How the curve looks ideally (given the graph scaling):
[image]

Gnarus-G, 2024-06-24:
@NeonNAX If this is still relevant to you.
You should be able to get more precision after the decimal point for your values now. Still not perfect, but much better.
Just reinstall, and re-enter your values.

Gnarus-G, 2024-06-25:
Graph should scale appropriately now.

closed 2024-06-25 by Gnarus-G

NeonNAX, 2024-06-28:
Tried the driver today.
Values are very accurate now and the grath looks to be in a readble area. Thanks for that.

To add to the device list:
- deathadder v3 wiered works
- g303 shroud edition does not
it fails at the binding for me

copy from terminal:
sudo maccel bind 7-3:1.0
[INFO] device to bind: Logitech USB Receiver
[INFO] unbinding from hid-generic
[INFO] device 7-3:1.0 is not bound to driver 'usbhid'
[INFO] binding to maccel
Error: failed to bind device '7-3:1.0' to driver 'maccel'

Caused by:
    Device or resource busy (os error 16)

(tried this to see if there was a problem with the bindall, thats why it was unbound from hid-generic)