Gnarus-G/maccel #14: After kernel update mouse stopped working.

ParanoIIa91, 2024-04-29:
After i reinstalled driver my mouse works again, but the problem is that applying same settings using "sudo maccel tui" does absolutely nothing?🤔

Gnarus-G, 2024-04-29:
There is kind of already an issue I opened about this #7 

It sound like the driver did not bind automatically, which can happen in this case. You can check with `lsusb -t | grep maccel`. You can also run `sudo maccel bindall` to do the binding that would have happened automattically.

If the maccel is bound to your mouse and you don't think it works, then screenshot your settings, and give me the output of `cat /sys/module/maccel/parameters/*`.

P.S. You shouldn't need to run the tui with `sudo`, otherwise something went wrong in the install process.

ParanoIIa91, 2024-04-30:
[image]
[image]
If i use tui without sudo, it wont let me change anything and "lsusb -t | grep maccel" command gives me nothing?

Gnarus-G, 2024-04-30:
Getting nothing means the driver isn't bound to any device. Run the bind all command I gave.

For the sudo on the tui, I guess you're fine with it. You probably didn't add yourself to the maccel group. So don't worry about it

ParanoIIa91, 2024-04-30:
I done sudo bind all command before, but this time it worked, thanks :)

Gnarus-G, 2024-05-01:
Great. I'll improve some of the documentation thanks to this issue.

closed 2024-05-01 by Gnarus-G