Gnarus-G/maccel #22: Not working for Razor Deathadder Elite V2

GENERALCHEEZITZ, 2024-07-28:
I know it's listed as one of the devices that works, but for some odd reason not for me. I tested this with another wired usb mouse I had lying around and it works. I don't get any kind of error when the device binds, it just says it binds and the mouse just stops working completely, no movement and no button input. I saw this as an issue with leetmouse, but the issue was closed with no conclusion https://github.com/systemofapwne/leetmouse/issues/41. Is my mouse just f*cked?

Gnarus-G, 2024-08-05:
Well to be fair, what is listed is not the Elite one.
I am working on a fix for these kind of issues. In fact you can help me if you would, try out the new implementation of the driver in this PR #23
It's a work in progress, but I need to know if I'm getting somewhere.

Gnarus-G, 2024-08-05:
Closing this since it's pretty much a duplicate issue from #9

closed 2024-08-05 by Gnarus-G