Gnarus-G/maccel #26: trouble installing on debian

estavrionto, 2024-08-17:
I tried the installation command but it did not work. The output is attached below.
[install_output.txt](https://github.com/user-attachments/files/16643594/install_output.txt)
I tried the uninstall command and tried to install it again, but did not work.

system info
[neofetch.txt](https://github.com/user-attachments/files/16643585/neofetch.txt)
apt list
[apt_list.txt](https://github.com/user-attachments/files/16643570/apt_list.txt)

Feel free to let me know if you need any more info.
Thanks

estavrionto, 2024-08-17:
better system info:
[neofetch.txt](https://github.com/user-attachments/files/16643602/neofetch.txt)

Gnarus-G, 2024-08-18:
```txt
insmod: ERROR: could not insert module /opt/maccel/driver/maccel.ko: Key was rejected by service
```
suggests that your kernel doesn't allow third party kernel modules from being installed. I don't know if that's debian's fault or not, but I can't do any about that. The module should work/is allowed on linux, and linux-zen kernels at least.

closed 2024-08-18 by Gnarus-G