Gnarus-G/maccel #28: maccel set all only sets multiplier

sweglord227, 2024-09-13:
Running on arch hyprland. i am using `maccel set all` to make keybinds to toggle mouse accel, but the command only changes the multiplier. accel, offset, and output-cap are unchanged.

Gnarus-G, 2024-09-13:
Just tried them, and it worked for me. I am going to need to more info, screenshots, video, etc...

What happens if you use:
```sh
maccel set param output-cap <number-here>
```

sweglord227, 2024-09-13:
```
~> maccel set param output-cap 2
Error: failed to write reset script

Caused by:
    Permission denied (os error 13)
```
i am apart of the maccel group as well. the weird part is that this does change the sens-mult, even tho i get the same error

Gnarus-G, 2024-09-13:
Something sus must have happened during your install. Uninstall and install again.
The reset scripts are backups for when you reboot, to maintain your settings. They should have been set to also be part of the `maccel` group.

sweglord227, 2024-09-13:
alright it works properly after reinstalling. thanks for your help

closed 2024-09-13 by sweglord227