Gnarus-G/maccel #30: Unable to install on CachyOS (Arch-based)

Vistaus, 2024-10-03:
I'm getting this error:

```
Fetching the maccel github repo
Cloning into '.'...
remote: Enumerating objects: 96, done.
remote: Counting objects: 100% (96/96), done.
remote: Compressing objects: 100% (88/88), done.
remote: Total 96 (delta 4), reused 50 (delta 2), pack-reused 0 (from 0)
Receiving objects: 100% (96/96), 2.20 MiB | 23.01 MiB/s, done.
Resolving deltas: 100% (4/4), done.

Installing the driver (kernel module)
rm -rf `pwd`/driver/.*.cmd `pwd`/driver/*.ko `pwd`/driver/*.mod `pwd`/driver/*.mod.* `pwd`/driver/*.symvers `pwd`/driver/*.order `pwd`/driver/*.o
rmmod: ERROR: Module maccel is not currently loaded
make: *** [Makefile:31: uninstall] Error 1
make CC=gcc EXTRA_CFLAGS= -C /lib/modules/`uname -r`/build M=`pwd`/driver
make[1]: Entering folder '/usr/lib/modules/6.11.0-5-cachyos-lto/build'
make[1]: *** No targets and Makefile found.  Cancelled.
make[1]: Leaving folder '/usr/lib/modules/6.11.0-5-cachyos-lto/build'
make: *** [Makefile:13: build] Error 2
```

The headers are installed:

```
ls /lib/modules
6.11.0-5-cachyos-lto/  6.11.1-2-cachyos-lto/

~/Downloads
❯ ls /lib/modules/6.11.1-2-cachyos-lto/
build/   modules.alias      modules.builtin            modules.builtin.bin      modules.dep      modules.devname  modules.softdep  modules.symbols.bin  pkgbase
kernel/  modules.alias.bin  modules.builtin.alias.bin  modules.builtin.modinfo  modules.dep.bin  modules.order    modules.symbols  modules.weakdep      vmlinuz
```

Distro: CachyOS
Kernel: Linux cachyos-HX90 6.11.0-5-cachyos-lto #1 SMP PREEMPT_DYNAMIC Fri, 20 Sep 2024 10:53:20 +0000 x86_64 GNU/Linux
DE: KDE Plasma 6.1.5 Wayland

How do I solve this issue?

pxlsec, 2024-12-03:
Hey, I'm new to kernel modules, but if I'm not mistaken `/usr/lib/modules/6.11.0-5-cachyos-lto/build` is a part of the kernel headers..
Could you ensure that the appropriate headers are installed? It should be something in the style of `cachyos-lto-headers` in your package manager.

Vistaus, 2024-12-03:
Hey, thanks for the reply. The cachyos-headers package is installed.

pxlsec, 2024-12-03:
How did you install?
Did you clone the repo and run `sudo ./install.sh` or did you `curl -fsSL https://www.maccel.org/install.sh | sudo sh`?
If you cloned the repo, what is the output of `ls -la driver/`. aka are all the files there, including the `Makefile`?
What kernel version are you using `uname -r`?
And lastly does ``/lib/modules/`uname -r`/build/Makefile`` exist?

dougg0k, 2025-01-26:
This fix worked on cachyos.

https://github.com/Gnarus-G/maccel/issues/34#issuecomment-2614638135

closed 2025-02-16 by Gnarus-G