Gnarus-G/maccel #31: Add other accel types

dougg0k, 2024-10-21:
Hi,

Can you add other accel types like Motivity?

And anisotropy, to alter X / Y of mouse independently?

Gnarus-G, 2024-10-24:
It's interesting to see that some people want this to be more like rawaccel. I'm iffy on wether I want feature parity with rawaccel, but I am not opposed to it. 
The issue is I don't yet understand the math past linear acceleration, and I don't have much time these days to dedicate to this project, but I will consider a pull request. For example Motivity is being worked on in #24

dougg0k, 2024-10-24:
Not much about being like rawaccel, but the functionality in itself.

Those that were used by me and brought me the results I was looking for, no other accel type did.

closed 2024-11-26 by dougg0k