Gnarus-G/maccel #32: Automate and publish to AUR

dougg0k, 2024-10-24:
https://github.com/marketplace/actions/publish-aur-package

https://wiki.archlinux.org/title/AUR_submission_guidelines

https://github.com/Gnarus-G/maccel/issues/30

Gnarus-G, 2024-10-24:
Thanks again for another good issue report. Thanks for the links.

Gnarus-G, 2025-03-13:
Would this make it so the post install scripts in the `maccel.install` https://github.com/Gnarus-G/maccel/blob/f0060556a336aaf262a5a544981045a5fb90ed08/maccel.install#L2 will work. 

As is, `makepkg -si` is really a three step process, including rebooting (potentially) your machine.

dougg0k, 2025-03-13:
What context is that?

Why, they dont currently work? You can see both yeetmouse files. That should somewhat answer. 

Or do you mean because of this? https://gitlab.archlinux.org/archlinux/packaging/packages/systemd/-/blob/204-2/PKGBUILD#L104

dougg0k, 2025-03-13:
And the context?

Since I just saw you already had all setup.

Gnarus-G, 2025-03-13:
Not sure what context in this case is, but I guess you mean if it's root. I can say that currently running `makepkg -si` doesn't run the .install hooks as root.

dougg0k, 2025-03-13:
https://github.com/Jguer/yay/issues/1026#issuecomment-532569850

https://www.reddit.com/r/linuxquestions/comments/13apj6i/force_makepkg_to_run_as_root/

greiner-petter, 2026-09-08:
Now that the AUR is allowing new registrations again, is this still being worked on?