Gnarus-G/maccel #37: Changable install paths for binaries(CLI), settings, logs and udev, and change default udev rule path.

MCPO-Spartan-117, 2024-12-05:
Distros will want to put settings in `/etc`, logs in `/var/log` and binaries in '/usr/bin' but the current setup requires a patch to do so.

Also the udev rules should default to `/usr/local/lib/udev/rules.d` instead, see `man 7 udev`.

Gnarus-G, 2025-01-26:
Do you have an idea for how you want those options to be setup?

dougg0k, 2025-02-14:
Not necessarily `/usr/local/lib/udev/rules.d/`, at least here on a arch distro, it's usually added to `/usr/lib/udev`. The suggested path does not exist.

Gnarus-G, 2025-02-17:
I'm not convinced this is that important then.

closed 2025-02-17 by Gnarus-G

MCPO-Spartan-117, 2025-02-22:
> Not necessarily `/usr/local/lib/udev/rules.d/`, at least here on a arch distro, it's usually added to `/usr/lib/udev`. The suggested path does not exist.

Almost like it is a directory for admins to use and not the distro.

dougg0k, 2025-02-22:
That is also correct. https://man7.org/linux/man-pages/man7/hier.7.html

Gnarus-G, 2025-02-22:
If this is just about the default path of the udev rules then I'm good with it. I think the issue title is asking for more but I'll accept a PR for better defaults. I do want follow conventions but I don't to complicate the install process too much.