Gnarus-G/maccel #39: Mouse matching is too strict

gnusenpai, 2024-12-08:
I have a weird mouse (Logitech MX Ergo) that doesn't have the usual USBMOUSE HID type.

Doing the really naive thing by changing:
https://github.com/Gnarus-G/maccel/blob/7a9dd6a61922a48dd8369beebb62d8c916dde6d1/driver/input_handler.h#L80
to:
```c
  return true;
```
seems to work.

I haven't noticed any adverse effects, but I don't have a ton of USB devices to test.

Gnarus-G, 2025-01-26:
Maybe that's allright since the main driver doesn't "hijack" your device, but I don't know how I feel about that as a general solution. I'll consider it when I decide to throw away the usbmouse driver that binds.

Gnarus-G, 2025-02-17:
Yep. Removed it. You should be good now without needing to patch it yourself.

closed 2025-02-17 by Gnarus-G