Gnarus-G/maccel #42: Cooperation with AndyFilter's Leetmouse fork

owosoup, 2024-12-20:
Maybe working with the [this project](https://github.com/AndyFilter/YeetMouse) can help. They have a working GUI and are testing mouse snapping. I'd say it's better to share progress, instead of trying to do the same thing separately.

MCPO-Spartan-117, 2024-12-20:
Looks like they are much farther ahead and might be going to use `input_handlers` in a better way.
https://github.com/AndyFilter/YeetMouse/pull/5

owosoup, 2024-12-20:
Do you plan to keep developing maccel further if YeetMouse is already ongoing? Closing this to keep it clean.

closed 2024-12-20 by owosoup

Gnarus-G, 2024-12-20:
This is very cool, thx for sharing.

Gnarus-G, 2024-12-20:
> Do you plan to keep developing maccel further if YeetMouse is already ongoing? Closing this to keep it clean.

I'll find some time to try theirs out, maybe talk to them if possible. There's not much innovation here on my part aside from maybe the input_handler and tui, so maybe theirs is better and I retire it, idk yet. Once I merged the current PRs I'll see I feel about the current state of maccel.

MCPO-Spartan-117, 2024-12-22:
The repo dev does seem to slam a bunch of stuff in one commit and often does a lot more than what the commit name suggests, implemented 6 different accel modes in the first commit but says nothing about it in the commit, putting this all in one commit definitely makes implementing more later harder, for others at least.