Gnarus-G/maccel #43: Install error on Debian 12

arut-plus, 2025-01-08:
```
➜ cat /proc/version
Linux version 6.1.0-28-amd64 (debian-kernel@lists.debian.org) (gcc-12 (Debian 12.2.0-14) 12.2
.0, GNU ld (GNU Binutils for Debian) 2.40) #1 SMP PREEMPT_DYNAMIC Debian 6.1.119-1 (2024-11-2
2)
➜ gcc --version
gcc (Debian 12.2.0-14) 12.2.0
➜ rustup --version
rustup 1.27.1 (54dd3d00f 2024-04-24)
info: This is the version for the rustup toolchain manager, not the rustc compiler.
info: The currently active `rustc` version is `rustc 1.83.0 (90b35a623 2024-11-26)`
➜ cargo --version
cargo 1.83.0 (5ffbef321 2024-10-29)
```
Installing:
```
➜ curl -fsSL https://www.maccel.org/install.sh | sudo -E env "PATH=$PATH" sh
sh: 61: [[: not found

Fetching the maccel github repo
sh: 51: [[: not found
Cloning into '.'...
remote: Enumerating objects: 97, done.
remote: Counting objects: 100% (97/97), done.
remote: Compressing objects: 100% (89/89), done.
remote: Total 97 (delta 4), reused 57 (delta 2), pack-reused 0 (from 0)
Receiving objects: 100% (97/97), 2.20 MiB | 7.33 MiB/s, done.
Resolving deltas: 100% (4/4), done.

Installing the driver (kernel module)
rm -rf `pwd`/driver/.*.cmd `pwd`/driver/*.ko `pwd`/driver/*.mod `pwd`/driver/*.mod.* `pwd`/driver/*.symvers `pwd`/driver/*.order `pwd`/driver/*.o
cargo clean --manifest-path=cli/Cargo.toml
     Removed 0 files
cargo clean --manifest-path=cli/usbmouse/Cargo.toml
     Removed 0 files
rmmod: ERROR: Module maccel is not currently loaded
make: *** [Makefile:35: uninstall] Error 1
sh: 88: [[: not found
make CC=gcc EXTRA_CFLAGS= -C /lib/modules/`uname -r`/build M=`pwd`/driver
make[1]: Entering directory '/usr/src/linux-headers-6.1.0-28-amd64'
  CC [M]  /opt/maccel/driver/maccel.o
In file included from /opt/maccel/driver/././accelk.h:4,
                 from /opt/maccel/driver/./input_handler.h:1,
                 from /opt/maccel/driver/maccel.c:1:
/opt/maccel/driver/././accel.h: In function ‘sensitivity’:
/opt/maccel/driver/././accel.h:29: warning: ignoring ‘#pragma clang diagnostic’ [-Wunknown-pragmas]
   29 |   dbg("accel    %s", fixedpt_cstr(param_accel, 6));
      |
/opt/maccel/driver/././accel.h:29: warning: ignoring ‘#pragma clang diagnostic’ [-Wunknown-pragmas]
/opt/maccel/driver/././accel.h:29: warning: ignoring ‘#pragma clang diagnostic’ [-Wunknown-pragmas]
/opt/maccel/driver/././accel.h:39: warning: ignoring ‘#pragma clang diagnostic’ [-Wunknown-pragmas]
   39 |   dbg("sens     %s", fixedpt_cstr(sens, 6));
      |
/opt/maccel/driver/././accel.h:39: warning: ignoring ‘#pragma clang diagnostic’ [-Wunknown-pragmas]
/opt/maccel/driver/././accel.h:39: warning: ignoring ‘#pragma clang diagnostic’ [-Wunknown-pragmas]
/opt/maccel/driver/././accel.h:40: warning: ignoring ‘#pragma clang diagnostic’ [-Wunknown-pragmas]
   40 |   dbg("sens cap %s", fixedpt_cstr(output_cap, 6));
      |
/opt/maccel/driver/././accel.h:40: warning: ignoring ‘#pragma clang diagnostic’ [-Wunknown-pragmas]
/opt/maccel/driver/././accel.h:40: warning: ignoring ‘#pragma clang diagnostic’ [-Wunknown-pragmas]
/opt/maccel/driver/././accel.h: In function ‘input_speed’:
/opt/maccel/driver/././accel.h:54: warning: ignoring ‘#pragma clang diagnostic’ [-Wunknown-pragmas]
   54 |   dbg("distance (in)              %s", fixedpt_cstr(distance, 6));
      |
/opt/maccel/driver/././accel.h:54: warning: ignoring ‘#pragma clang diagnostic’ [-Wunknown-pragmas]
/opt/maccel/driver/././accel.h:54: warning: ignoring ‘#pragma clang diagnostic’ [-Wunknown-pragmas]
/opt/maccel/driver/././accel.h:59: warning: ignoring ‘#pragma clang diagnostic’ [-Wunknown-pragmas]
   59 |       fixedpt_cstr(fixedpt_fromint(polling_interval), 6));
      |
/opt/maccel/driver/././accel.h:59: warning: ignoring ‘#pragma clang diagnostic’ [-Wunknown-pragmas]
[... 193 lines cut ...]
      |                                     ^~~~~~~~~~~
/opt/maccel/driver/././fixedptc.h:112:35: note: in definition of macro ‘fixedpt_add’
  112 | #define fixedpt_add(A, B) ((A) + (B))
      |                                   ^
/opt/maccel/driver/././accel.h:29:22: warning: ‘fixedpt_cstr’ is static but used in inline function ‘sensitivity’ which is not static
   29 |   dbg("accel    %s", fixedpt_cstr(param_accel, 6));
      |                      ^~~~~~~~~~~~
/usr/src/linux-headers-6.1.0-28-common/include/linux/printk.h:429:33: note: in definition of macro ‘printk_index_wrap’
  429 |                 _p_func(_fmt, ##__VA_ARGS__);                           \
      |                                 ^~~~~~~~~~~
/opt/maccel/driver/././dbg.h:20:7: note: in expansion of macro ‘printk’
   20 |       printk(KERN_INFO "%s:%d:%s(): " #fmt "\n", __FILE__, __LINE__, __func__, \
      |       ^~~~~~
/opt/maccel/driver/././dbg.h:9:23: note: in expansion of macro ‘dbg_k’
    9 | #define dbg(fmt, ...) dbg_k(fmt, __VA_ARGS__)
      |                       ^~~~~
/opt/maccel/driver/././accel.h:29:3: note: in expansion of macro ‘dbg’
   29 |   dbg("accel    %s", fixedpt_cstr(param_accel, 6));
      |   ^~~
cc1: some warnings being treated as errors
make[2]: *** [/usr/src/linux-headers-6.1.0-28-common/scripts/Makefile.build:255: /opt/maccel/driver/maccel.o] Error 1
make[1]: *** [/usr/src/linux-headers-6.1.0-28-common/Makefile:2034: /opt/maccel/driver] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-6.1.0-28-amd64'
make: *** [Makefile:17: build] Error 2
```

arut-plus, 2025-01-11:
?

Gnarus-G, 2025-01-26:
Can you try now?

arut-plus, 2025-01-26:
> Can you try now?

## Versions
```
➜ cat /proc/version
Linux version 6.1.0-28-amd64 (debian-kernel@lists.debian.org) (gcc-12 (Debian 12.2.0-14) 12.2.0, GNU ld (GNU Binutils for Debian) 2.40) #1 SMP PREEMPT_DYNAMIC Debian 6.1.119-1 (2024-11-22)
➜ gcc --version
gcc (Debian 12.2.0-14) 12.2.0
➜ rustup --version
rustup 1.27.1 (54dd3d00f 2024-04-24)
info: The currently active `rustc` version is `rustc 1.83.0 (90b35a623 2024-11-26)`
➜ cargo --version
cargo 1.83.0 (5ffbef321 2024-10-29)
```

## Installation
```
➜ curl -fsSL https://www.maccel.org/install.sh | sudo -E env "PATH=$PATH" sh
sh: 66: [[: not found

Fetching the maccel github repo
sh: 56: [[: not found
Cloning into '.'...
remote: Enumerating objects: 101, done.
remote: Counting objects: 100% (101/101), done.
remote: Compressing objects: 100% (93/93), done.
remote: Total 101 (delta 4), reused 59 (delta 1), pack-reused 0 (from 0)
Receiving objects: 100% (101/101), 2.21 MiB | 6.84 MiB/s, done.
Resolving deltas: 100% (4/4), done.

Installing the driver (kernel module)
Sign command: /usr/lib/linux-kbuild-6.1/scripts/sign-file
Signing key: /var/lib/dkms/mok.key
Public certificate (MOK): /var/lib/dkms/mok.pub
Creating symlink /var/lib/dkms/maccel/0.2.1/source -> /usr/src/maccel-0.2.1

Building module:
Cleaning build area...
make -j16 KERNELRELEASE=6.1.0-28-amd64 KVER=6.1.0-28-amd64...(bad exit status: 2)
Error! Bad return status for module build on kernel: 6.1.0-28-amd64 (x86_64)
Consult /var/lib/dkms/maccel/0.2.1/build/make.log for more information.
```

## Log

I'm attaching a log.
`/var/lib/dkms/maccel/0.2.1/build/make.log`: [make.log](https://github.com/user-attachments/files/18552025/make.log)

Gnarus-G, 2025-01-26:
What shell are using for `[[` not to be working?

Gnarus-G, 2025-01-26:
Maybe your `gcc` version is just too low.

arut-plus, 2025-01-27:
> What shell are using for `[[` not to be working?

The syntax with **two square brackets** is **not** POSIX-compliant, so **sh** does **not** support it.
And the [README](https://github.com/Gnarus-G/maccel?tab=readme-ov-file#install) specifies exactly that command for installation.

Of course, I also ran with **bash**, but that only got rid of the warnings about unsupported two square brackets, but it didn't solve the installation problem itself - the installation error is the same.

---

> Maybe your `gcc` version is just too low.

The version of gcc, as I understand it, should correspond to the version with which the kernel was built.

Gnarus-G, 2025-01-27:
Didn't know that about the brackets, weird that it always worked with sh for me.

Yes the main issue seems that we don't backwards compat with some older version gcc.

Thanks for the report.

closed 2025-02-17 by Gnarus-G

arut-plus, 2025-02-23:
Thanks for trying and for the tip. 💟 
Anyway, I've already switched to another distribution (nixos).

_P.s. I caught a kernel panic back then on debian due to trying to update gcc... 😄_