Gnarus-G/maccel #45: Bug: Slower movement in higher mouse hertz

dougg0k, 2025-01-27:
Hi,

Just found a bug related to hz in the mouse.

I was using the cable which only get to about 1000hz, the graph and sensitivity were quickly changing.

Now, after changing to 4000hz, all seem slowed down in the sensitivity of the graph. Might require to detect change in the hz or have a way to input your current hz, so you can do the calculations based on it.

It's like with the same or more mouse movement, the dots in the graph move less. From simply going from 1000 to 4000hz.

Changing the offset to half, got close to what it was. But it feels weird, as it takes longer reach the accel. Since the dots takes longer to move. Feels inconsistent.

---

This need to be fixed, otherwise, it's unusable for me at least.

Gnarus-G, 2025-01-27:
To be clear, you're also feeling a different in the mouse movement, right? You're not just reporting a visual bug.

dougg0k, 2025-01-27:
Yes.

dougg0k, 2025-01-27:
In addition, which is unrelated to the issue.

It's missing other curves option which would make the transition from current to accel, more smooth and continous. Not, like when it's in the default speed, then suddenly very fast. 

There are probably ways to make the current better when changing the numbers, but still there are results like this.

dougg0k, 2025-01-29:
Tested a bit more today, very much noticeable. It's like with 1000hz, it's snappier and responsive. And 4000 just slower version.

dougg0k, 2025-01-31:
Video showing the difference between the two.

https://github.com/user-attachments/assets/ecf0469e-9706-4ee8-b14c-463b8f24a318

Gnarus-G, 2025-02-03:
> Unrelated to the issue. In accel input, if I set a value and close the tui, after opening it again, it returns to a bigger float. Might be a bug.
> 

Not a bug per se. Just a quirk of using fixed point notation for the values which is just an approximation

> Another point I noticed today, besides all the other devices I had pointer accel set to none on kde, your maccel virtual device was added with the standard mode. There was a difference. But the bug are still there.
> 
> Perhaps you should add to the readme that the DE might be stupid, so, be sure to check.
Interesting. If you make a pull request with an update to the README I'll accept it.

dougg0k, 2025-02-14:
Today I tried a different one, definetely makes a difference in accuracy. With 4000hz, your accuracy are more rigid, while in 1000hz it becomes better. But wireless are still better, and wireless are always 4000hz.

Can this be fixed?

dougg0k, 2025-02-17:
I just tried your update. Virtual driver is gone, seems good.

This issue still remains the same.

Gnarus-G, 2025-02-17:
- The UI currently always assumes a 1ms polling interval (1000Hz polling rate)
- The driver guards against anything less than a millisecond.
- To help me confirm somethings, run the driver in debug mode and send me an output containing logs about `polling_interval`
- - After cloning the repo, cd into it and,
- - Run these command
 ```sh 
make uninstall; 
make debug_install;
 ```
Note: `make refresh` will get you back to the non-debug driver

- Lastly, what mouse is this you have that's doing 4000Hz? Share a link please.

dougg0k, 2025-02-17:
Why does it guard against it?

```
❯ make debug_install
/usr/bin/make CC=gcc EXTRA_CFLAGS=-DDEBUG -C `pwd`/driver
make[1]: Entering directory '/home/user/Downloads/maccel/driver'
/usr/bin/make -C /lib/modules/6.13.2-2-cachyos/build M=/home/user/Downloads/maccel/driver
make[2]: Entering directory '/usr/lib/modules/6.13.2-2-cachyos/build'
make[3]: Entering directory '/home/user/Downloads/maccel/driver'
warning: the compiler differs from the one used to build the kernel
  The kernel was built by: clang version 19.1.7
  You are using:           gcc (GCC) 14.2.1 20250207
  CC [M]  maccel.o
gcc: error: unrecognized command-line option ‘-mretpoline-external-thunk’
gcc: error: unrecognized command-line option ‘-fsplit-lto-unit’
gcc: error: unrecognized command-line option ‘-fdebug-info-for-profiling’
gcc: error: unrecognized command-line option ‘-mllvm’
gcc: error: unrecognized command-line option ‘-mllvm’
gcc: error: unrecognized command-line option ‘-improved-fs-discriminator=true’
gcc: error: unrecognized command-line option ‘-fbasic-block-sections=labels’
make[5]: *** [/usr/lib/modules/6.13.2-2-cachyos/build/scripts/Makefile.build:196: maccel.o] Error 1
make[4]: *** [/usr/lib/modules/6.13.2-2-cachyos/build/Makefile:1997: .] Error 2
make[3]: *** [/usr/lib/modules/6.13.2-2-cachyos/build/Makefile:251: __sub-make] Error 2
make[3]: Leaving directory '/home/user/Downloads/maccel/driver'
make[2]: *** [Makefile:251: __sub-make] Error 2
make[2]: Leaving directory '/usr/lib/modules/6.13.2-2-cachyos/build'
make[1]: *** [Makefile:7: build] Error 2
make[1]: Leaving directory '/home/user/Downloads/maccel/driver'
make: *** [Makefile:16: build] Error 2
```

It's a Lamzu 4k mini.

Gnarus-G, 2025-02-17:
> Why does it guard against it?
> 
Idk, carried over from when I ported leetmouse. I probably don't need it.
> ```
> ❯ make debug_install
> /usr/bin/make CC=gcc EXTRA_CFLAGS=-DDEBUG -C `pwd`/driver
> make[1]: Entering directory '/maccel/driver'
> /usr/bin/make -C /lib/modules/6.13.2-2-cachyos/build M=/maccel/driver
> make[2]: Entering directory '/usr/lib/modules/6.13.2-2-cachyos/build'
> make[3]: Entering directory '/maccel/driver'
> warning: the compiler differs from the one used to build the kernel
>   The kernel was built by: clang version 19.1.7
>   You are using:           gcc (GCC) 14.2.1 20250207
>   CC [M]  maccel.o
> gcc: error: unrecognized command-line option ‘-mretpoline-external-thunk’
> gcc: error: unrecognized command-line option ‘-fsplit-lto-unit’
> gcc: error: unrecognized command-line option ‘-fdebug-info-for-profiling’
> gcc: error: unrecognized command-line option ‘-mllvm’
> gcc: error: unrecognized command-line option ‘-mllvm’
> gcc: error: unrecognized command-line option ‘-improved-fs-discriminator=true’
> gcc: error: unrecognized command-line option ‘-fbasic-block-sections=labels’
> make[5]: *** [/usr/lib/modules/6.13.2-2-cachyos/build/scripts/Makefile.build:196: maccel.o] Error 1
> make[4]: *** [/usr/lib/modules/6.13.2-2-cachyos/build/Makefile:1997: .] Error 2
> make[3]: *** [/usr/lib/modules/6.13.2-2-cachyos/build/Makefile:251: __sub-make] Error 2
> make[3]: Leaving directory /maccel/driver'
> make[2]: *** [Makefile:251: __sub-make] Error 2
> make[2]: Leaving directory '/usr/lib/modules/6.13.2-2-cachyos/build'
> make[1]: *** [Makefile:7: build] Error 2
> make[1]: Leaving directory '/maccel/driver'
> make: *** [Makefile:16: build] Error 2
> ```
> 
Weird. `export CC=clang;` and then run the make commands again.

> It's a Lamzu 4k mini. (select the name, right click, search google for...)
> 
lul, a little sassy, aren't we.

dougg0k, 2025-02-17:
Worked. 

Where that output are being sent? How to get it?

Gnarus-G, 2025-02-17:
> Where that log output are being sent? How to get it?
> 
`dmesg -w`

> Why do you ask commands to be typed by the user when you are already asking for sudo on install?
> 
Irrelevant to the current issue, but it's about following linux conventions, at least for parts of the process that are either, not absolutely necessary, or too important not to expose to the user to decide.

dougg0k, 2025-02-17:
Irrelevant but part. 

You could specify the comands and reason to run, by provinding a question, to proceed in executing all of them or not (y/n).

dougg0k, 2025-02-17:
https://github.com/user-attachments/assets/2cb56d44-3e90-4496-95a6-22b20fb37114

Nothing.

Gnarus-G, 2025-02-17:
Grep for 'polling interval' instead. Or better yet don't grep at all and just show me the whole thing.

dougg0k, 2025-02-17:
My bad.

https://www.dropbox.com/scl/fi/xjw71icywvyyv8qvp6csr/2025-02-17_15-51-36.mp4?rlkey=42yv5fb0uf27zyoru8qa4kyi1&st=23vhw5tz&dl=0

dougg0k, 2025-02-18:
Why accelerate are using time rather than the actual pooling rate from the device, faster perhaps? Or is it me misunderstanding?

https://github.com/Gnarus-G/maccel/blob/main/driver/accelk.h#L12

If not, maybe this could be of use. https://github.com/marcoster/evhz/blob/master/evhz.c

Gnarus-G, 2025-02-21:
The polling rate is a static value as the minimum interval between each event by the mouse. The real time is more important and necessary to know how fast you move your mouse.

Anyway, try this PR #53, and let me know how it fares for you.

After you pull the latest.
```sh
git checkout feat/support-high-polling-rate
make uninstall;
make install; # do 'install_debug' if you think there's a problem, to see the logs in dmesg
make install_cli; # This builds and installs the cli in your /usr/local/bin
````
Make sure you remember your values, they will be forgotten/clobbered.
```sh
maccel get all; # lists your current parameters
```

dougg0k, 2025-02-21:
I tested, first, without a restart, values went through, but the bar on the tui didint move at all, in any hz.

After, I tried to remove the cable and use the dongle, mouse cursor stopped working all together, didnt move anymore.

I did a reboot on the system. At first, it didnt boot, there was a blue line. I did a hard reboot, then it went through. 

But the mouse were uncontrollable in both cable or dongle.

Then I uninstalled, and the mouse worked correctly right after.

So, perhaps you should test first in your machine if works fine on your 1000hz.

Gnarus-G, 2025-02-21:
Weird, yeah it worked for me, that's why I wanted you to try it.

Just fixed the dot not moving in the clip as well. You should know this update will not change the speed of the dot in the UI. That will still be biased to 1000Hz.

Ok thanks for trying it so quickly.

dougg0k, 2025-02-21:
Sure.

But, what is the point of it then? Because the UI is the representation of not getting accelerated. It's not about the UI, but getting the correct acceleration.

Gnarus-G, 2025-02-21:
The point is you will get the correct acceleration, but the UI will not show it. Going by your last sentence, it looks like you're fine with that.

dougg0k, 2025-02-21:
So, if I set something in 1000hz where the UI are suppose to work, and change to a higher hz, it will work exact the same, accel wise?

Gnarus-G, 2025-02-21:
The input speed indicator will work the exact same. 

The *actual** acceleration will be <ins>different</ins>, and you will <ins>feel it</ins>.

dougg0k, 2025-02-21:
Is your `make uninstall`, temporary?

Yesterday I run the comand to stop the broken uncontrollable cursor, and today it was back and had to run the command again.

Gnarus-G, 2025-02-21:
I don't even know. Between the dkms install we do and this, there's probably some conflict. These commands are intended just for dev and testing.

Anyway, a clean uninstall will be with the `uninstall` script. So try that to be sure.

```sh
./uninstall.sh # from the root of the repo
```

dougg0k, 2025-02-22:
Right.

I installed your new changes. Mouse at 4khz stopped again, but the OS didnt freeze. 

Then I looked up the values through maccel tui, and sensitivity went from 1 to 0.0...., and I changed back to 1, it seems that accel might be there, but the graph still show as before (broken like).

dougg0k, 2025-02-22:
Before reboot, I set the new values, same as I did before, it worked.

But I did a reboot to check whether it would still be broken, it it was the all the values were wrong except for accel value which remained set.

Sensitivity went to 0.00002 with a write error if I tried to set back to 1. I didnt check the rest.

Gnarus-G, 2025-02-22:
You have to install the new CLI as well.

dougg0k, 2025-02-22:
Yes

dougg0k, 2025-02-22:
See what I just found when updating the kernel.

```
( 6/15) Install DKMS modules
==> dkms install --no-depmod maccel/0.3.1 -k 6.13.4-2-cachyos
==> dkms install --no-depmod maccel/0.2.1 -k 6.13.4-2-cachyos
...
```

You might need to improve your uninstall process, or overwrite process.

dougg0k, 2025-02-22:
After a restart, I tried doing a `make install` on the branch. But got this error.

```
...
changed ownership of '/sys/module/maccel/parameters/OFFSET' from root:root to :maccel
changed ownership of '/sys/module/maccel/parameters/OUTPUT_CAP' from root:root to :maccel
changed ownership of '/sys/module/maccel/parameters/SENS_MULT' from root:root to :maccel
chown: cannot access '/var/opt/maccel/resets/*': No such file or directory
failed to change ownership of '/var/opt/maccel/resets/*' to :maccel
changed ownership of '/dev/maccel' from root:root to :maccel
make: *** [Makefile:19: install] Error 1
```

Maybe it's because that one commit behind?

dougg0k, 2025-02-22:
After the complete uninstall, I did install through `curl -fsSL https://www.maccel.org/install.sh | sudo sh` on v0.3.2.

After, I did the steps to use the changes from the branch. Did a Reboot.

So far, no problems, except that it is like before, no accel on 4khz.

Gnarus-G, 2025-02-23:
I was making a lot changes to the install/uninstall process because I also hit those issues. It's much better now.

You don't even feel a tactile difference? Like I said before, you won't *see* it. 
At some point later, I'll add a field like rawaccel to input an arbitrary polling rate.

Anyway I ordered an 8k rate mouse, so I'll be able to test myself.

dougg0k, 2025-02-23:
I did when there was that problem before, but not now. With the same movement I can go through both monitors with accel on 1khz, yet on the 4khz, it's barely one monitor.

Gnarus-G, 2025-02-23:
Go to debug mode, send me a screenshot/sample of your dmesg. And one of your parameters `maccel get all` will do as well.

Gnarus-G, 2025-02-23:
Which branch are you using? Just to double check.

dougg0k, 2025-02-23:
```
~/maccel on  feat/support-high-polling-rate
❯ maccel get all                                                                                 21:44:06
Sens-Multiplier:        1.0
Accel:          0.1000061035
Offset:         10.0
Output-Cap:     2.0
```

Nothing relevant on dmesg.

Gnarus-G, 2025-02-23:
Maybe your dkms version is stuck in an older install. 

Just don't worry for now. I just got the new mouse, I'll sort things out.

dougg0k, 2025-02-23:
Not anymore.

```
❯ dkms status                                                                                                                                                                                                                                                                      
...
maccel/0.3.2, 6.13.4-2-cachyos, x86_64: installed
...
```

dougg0k, 2025-02-23:
```
❯ dkms status | grep maccel                                                                                                                                                                               
maccel/0.4.0, 6.13.4-2-cachyos, x86_64: installed

~/maccel on  feat/support-high-polling-rate
❯ make install_debug                                                                                                                                                                                        ...
changed ownership of '/sys/module/maccel/parameters/SENS_MULT' from root:root to :maccel
chown: cannot access '/var/opt/maccel/resets/*': No such file or directory
failed to change ownership of '/var/opt/maccel/resets/*' to :maccel
changed ownership of '/dev/maccel' from root:root to :maccel
make: *** [Makefile:19: install] Error 1
```

Gnarus-G, 2025-02-23:
How do you change your polling rate? Is there a super easy way or is it just the `usbhid` mousepoll option I keep seeing while I'm googling.

You can try the PR again now.

dougg0k, 2025-02-23:
Last time I tried mousepoll option both keyboard and mouse stopped working.

Find a tool that support whatever mouse you have.

dougg0k, 2025-02-23:
Same error.

Gnarus-G, 2025-02-23:
Same error on version 0.4.1?

dougg0k, 2025-02-23:
? there is no 0.4.1

Gnarus-G, 2025-02-23:
Which install command(s) are you using?

dougg0k, 2025-02-23:
Oh, you mean you bump the PR.

```
~/maccel on  feat/support-high-polling-rate
❯ make uninstall

~/maccel on  feat/support-high-polling-rate
❯ make install_debug
```

Gnarus-G, 2025-02-23:
From now on, until I merge. Run this command.
```sh
curl  -fsSL https://www.maccel.org/install.sh | sudo BRANCH=feat/support-high-polling-rate BUILD_CLI_FROM_SOURCE=1 sh
```
Run a clean uninstall first too. Might as well.

dougg0k, 2025-02-23:
```
[14032.082578] ./././accelk.h:24:accelerate(): "ktime interval -> now (14032176229199) vs last_ktime (14032174227158), diff = 2002041ns"
[14032.082580] ./././accelk.h:37:accelerate(): "ktime interval -> converting to ns: 2002041ns -> 2.002040999ms"
[14032.082592] ./././accel.h:67:input_speed(): "time interval              2.002040999"
[14032.083577] ./././accelk.h:24:accelerate(): "ktime interval -> now (14032177228921) vs last_ktime (14032176229199), diff = 999722ns"
[14032.083579] ./././accelk.h:37:accelerate(): "ktime interval -> converting to ns: 999722ns -> 0.999721999ms"
[14032.083592] ./././accel.h:67:input_speed(): "time interval              0.999721999"
[14032.084577] ./././accelk.h:24:accelerate(): "ktime interval -> now (14032178228444) vs last_ktime (14032177228921), diff = 999523ns"
[14032.084579] ./././accelk.h:37:accelerate(): "ktime interval -> converting to ns: 999523ns -> 0.999522999ms"
[14032.084591] ./././accel.h:67:input_speed(): "time interval              0.999522999"
[14032.086576] ./././accelk.h:24:accelerate(): "ktime interval -> now (14032180227288) vs last_ktime (14032178228444), diff = 1998844ns"
[14032.086577] ./././accelk.h:37:accelerate(): "ktime interval -> converting to ns: 1998844ns -> 1.998843999ms"
[14032.086588] ./././accel.h:67:input_speed(): "time interval              1.998843999"
```

dougg0k, 2025-02-23:
Now it seems to be working. On both 1khz and 4khz. Except that the graph still only for 1khz.

dougg0k, 2025-02-23:
Maybe I am mistaken but there seem to be some slight smoothness added.

Gnarus-G, 2025-02-23:
What do you mean by that?

dougg0k, 2025-02-23:
I dont know, I might be mistaken, cursor seem less floaty.

dougg0k, 2025-02-23:
Regardless. At least the foundation seems decent now. Though the transition from non to accel with linear are still bad imo.

dougg0k, 2025-02-23:
Feel free to merge and close the issue if you want.

Gnarus-G, 2025-02-23:
You meant it was smoother? Yeah that's probably due to having more precision on calculating the time intervals and removing the arbitrary caps I mentioned many messages ago.

Cool. I'm still trying to find a tool to set my higher polling rate myself.

closed 2025-02-23 by Gnarus-G

dougg0k, 2025-02-23:
What is your mouse?

Gnarus-G, 2025-02-23:
Razer DeathAdder V3 Pro.

I figured Razer has good Linux support with openrazer right? Not good enough I guess. Smh.

[4 more replies]

PR #53 (2025-02-23): Feat/support high polling rate
Fixes #45 
Depends on, though not strictly, PR #52 

#### To try the PR.

After you clone or pull the latest.
```sh
git checkout feat/support-high-polling-rate
make uninstall;
make install; # do 'install_debug' if you think there's a problem, to see the logs in dmesg
make install_cli; # This builds and installs the cli in your /usr/local/bin
````
Make sure you remember your values, they will be forgotten/clobbered.
```sh
maccel get all; # lists your current parameters
```