Gnarus-G/maccel #54: Add basic anisotropy support

dougg0k, 2025-02-23:
Hi,

Could you add basic anisotropy functionality?

Separation of X/Y sensitivity into two inputs. That's all.

Thanks.

Gnarus-G, 2025-02-23:
If it's all the same to you, then use Yeetmouse, https://github.com/AndyFilter/YeetMouse, it has a GUI, and a lot more curves. It now also supports any mouse that linux already supports since they ported the [input handler] method I have been using to Yeetmouse.

dougg0k, 2025-02-23:
Your cli are less bloated, but also less featured. If you want to improve yours, then I suppose this request are still valid.

dougg0k, 2025-02-24:
https://github.com/AndyFilter/YeetMouse/commit/8f6d77c2ee7a3289848a1f0a70858f39d4bb9403

Gnarus-G, 2025-02-25:
Yeetmouse's implementation seems to be such that the sens multiplier for the y axis Is a multiplier on top of the speed for the x axis. So a sens y of 2 is actually 2*sens_x. 

Is this the behavior you're talking about? 

My first thought was that you wanted By component anisotropy (to use rawaccel terms). I guess?

dougg0k, 2025-02-25:
Their impl are weird, if you can do better, works for me.

Gnarus-G, 2025-03-02:
Features in #55 solves this.

dougg0k, 2025-03-02:
Nice, let me know when you merge and release.

closed 2025-03-02 by dougg0k

dougg0k, 2025-03-02:
Just saw and tried it. 

Seems to be working. 

A few points, your install script doesnt seem to overwrite the current version and instead install a new one. And something that doesnt really matter, your cli version are not in sync.

Gnarus-G, 2025-03-02:
1. No it should do, you must have jumped on it too quickly and the newest executable hadn't released yet?
2. I don't want them to be synced up.

dougg0k, 2025-03-02:
Nah, both 0.4.1 and 0.4.3 were installed, at least shown on `dkms status`.

Gnarus-G, 2025-03-02:
Hmmm. We'll there would only be one version of the driver `modprobe`'d. Not sure dkms defines as "[installed]"
If it worked then you got the right version.

dougg0k, 2025-03-02:
I didnt test, I uninstalled all before reinstalling latest. But ok.