Gnarus-G/maccel #57: So I've been making a GUI...

sy3c4ll, 2025-03-05:
First off, thanks for creating this amazing project. I was almost ready to give up on mouse accel until I found this.

Also, I just have to say up front that I love your TUI as well. It's uncluttered, simple and highly usable. But I found using codepoints to plot the graph led to a loss of clarity, albeit miniscule.

So for the last week or two I've been [drafting](https://github.com/sy3c4ll/maccel-gui-prot) a **prototype** of how a GUI for maccel might look like, using Rust and Iced.

Here's how it's looking so far:

[image]

Like I said, so far it's only a prototype that works on dummy data. And to proceed from here I need your input.

See, I really doubt you weren't skilled enough to make such a simple GUI by yourself. Which means you may not _want_ a GUI in your project, and for good reason:

- It pulls in extra dependencies (iced)
- It breaks convention, many Linux mouse accel solutions have out-of-tree GUIs
- The TUI already does basically the same thing

So while I'm wishing my GUI could be upstreamed here and put under the care of someone more responsible than me, I'm equally fine with going my own way and hosting it out-of-tree.

So please help me decide! You could either:

1. Allow me to contribute a GUI directly to the repo. I'll be forking maccel and adapting my prototype to work with your existing library, altogether it'll most likely be a two-file addition. I'm also thinking of feature gating the GUI so the dependency is optional.
2. Reject this proposal. I'll be adding interop with your driver to my prototype to make it standalone, and keep hosting under my name. Probably it won't gain traction and I'll get tired of maintaining it eventually, I really can't be trusted with my own projects.

Thanks again, and RSVP.

Gnarus-G, 2025-03-05:
At the end of the day, the main limitation for me is time. Still, if you want to chat more, or work together to make a maccel GUI, than maybe you'd like to join the discord [server](https://discord.gg/SsPZhSKh)

sy3c4ll, 2025-03-06:
Wow that was prompt! I fully understand where you're coming from; if we're gonna add extra frontends, a little restructuring could really go a long way. In fact, I think I may have actually been hoping for something like this, so duplicate code between CLI, TUI and GUI can be reduced to a minimum.

As for Tauri, I'm afraid I'm in the opposite situation as you - I'm hot garbage on anything web, so I haven't tried Tauri thus unfortunately I am unable to offer my experiences there.

A while ago I tested out a few GUI solutions for Rust to find something to settle on - iced, egui and floem I think. I ultimately went with iced because I found the Elm architecture particularly pleasant to work with, and also because System76's new Cosmic DE uses it. Now I would be lying if I said I wasn't reconsidering my choices even a little bit, since 0.13 brought _massive_ (as in, nothing old works anymore) breaking changes and the developer's attitude seems quite... cavalier. Meanwhile I have hopes on translating this over to libcosmic for better theming and integration.

Please do take your time! In the meantime, I'll be adding integrations myself to make my Proof of Concept an actual Concept and make it work standalone. I still hope someday you'll be done with refactoring, and I'll be able to contribute.

Until then, I'll hop onto the Discord server. See you there!

sy3c4ll, 2025-03-06:
Also, feel free to close this issue if you think it should be closed! Or you could also keep it open as a tracking issue kind of thing, if you so wish. I dunno, I'll leave the decision to you!

dougg0k, 2025-03-13:
What is the point of GUI if you have TUI? For a simple tool.

Gnarus-G, 2025-03-13:
like I said: ~~plebes~~

dougg0k, 2025-03-13:
Just improve TUI if anything is needed.

Gnarus-G, 2025-03-13:
It's impossible to improve the TUI to fix the graph. That's the whole point. And it's optional, and it won't get any real love (dev time) from once it's done.

dougg0k, 2025-03-13:
What is there to fix there? Because it's dots? Lol

Gnarus-G, 2025-03-13:
@dougg0k Oh I see, I thought you wanted to understand why, but you're just talking shit. I get your sentiment, but you clearly don't have anything to add, so leave it alone.

dougg0k, 2025-03-13:
Wow, those were good points. 

What is there to add? It's a TUI. 

But here you go  https://github.com/tarkah/tickrs

dougg0k, 2025-03-17:
Anything to add?

You can just add mouse support to the TUI as something optional, that way you can also add desktop shortcut which would straight spawn a terminal with maccel and mouse support enabled. After all, all that there is available to do are set values and mode.

For such a simple tool, I dont see the point in wasting time on prettier chart or else. You could try to refine the chart to have a more clear line in certain situations, but only that.

Gnarus-G, 2025-03-18:
Obviously I agree with you about the TUI being good enough. You're also right that it could be a distraction because most features you want from a GUI you can get in TUI.

The best reasons we know of so far to have a GUI are:
- a graph that is much easier to scrutinize
- a place to dump any complicated UI features people might, which is me trying to read the future so NVM this.

Now

> could try to refine the chart to have a more > > clear line in certain situations.

That is a great point. It will be tricky but we could implement a zoom in/out on the TUI so I could use a lot more codepoints relative to the area.

I could also increase the number of data points overall to increase resolution (the limit factor for this so far has been performance)

Anyway I still feel that a lot people will prefer a GUI and I want to have that option eventually.

And improving the TUI when possible is definitely a higher priority.