Gnarus-G/maccel #58: DPI normalization

pxlsec, 2025-03-07:
Similar to libinput and rawaccel, we could normalize the dpi of the mouse to 1000 dpi.

It would also allow for high dpi mice to work at full resolution making curves with higher rates of change like "natural" more predictable at slower movements. This would also benefit portability by allowing transferring curves from one mouse to another and getting the exact same feeling.

From my understanding, implementing this would be as simple as scaling the input speed by (1000/dpi).

Gnarus-G, 2025-03-07:
Oh that's interesting. I didn't know to care about this. Now to just figure out what the dpi of a particular mouse is in Linux.

pxlsec, 2025-03-07:
Well,
I don't think the dpi is reported by the mouse. Libinput uses hwdb to query common mouse defaults but it's a hell to add an entry / edit an entry every time you change dpi.
I think it would be better to use the rawaccel approach of just having a multiplier. Or better yet make it per device with a config file #49; where you could have multiple devices listed using vendor/product id as listed by `lsusb`.

``` TOML
[Device.25a7:fa7b]
dpi = 1600
profile = "default"

[Device.25a7:fa7c]
dpi = 1600
profile = "default"

[Profile.default]
sensitivity = 1.0
type = "linear"
accel = 0.06
offset = 7.0
cap = 4.0
```

Gnarus-G, 2025-03-07:
Not totally familiar with the rawaccel approach.

Are you saying the dpi should be a user defined setting?

pxlsec, 2025-03-07:
> Not totally familiar with the rawaccel approach.

From what I remember its just a key/value pair in a text file, but it appears that it's made it into the gui since then. https://github.com/RawAccelOfficial/rawaccel/blob/master/doc/Guide.md#dpi-normalization

> Are you saying the dpi should be a user defined setting?

Yep, and you could always have a default value of 1000 dpi to make it do nothing as a default behavior.

closed 2025-03-09 by Gnarus-G