Gnarus-G/maccel #62: Add limit to ratio on synchronous

dougg0k, 2025-03-10:
Hi,

Can you add ratio limit to synchronous mode?

I was trying some values in attempt to find combination, one decent combination that I found had a problem, the ratio end were too high, so, if there were a way set a limit, that would help.  

[image]

Curve worked well for better accuracy, but it went too high.

Gnarus-G, 2025-03-10:
I don't understand what you're asking. What is happening? How are you getting it to happen? What would you like to happen instead?

dougg0k, 2025-03-10:
? 

Be able to set a limit, have an additional input. 

e.g., Let's say that horizontal line in the screenshot, stops at 8.0~, if I could limit that to different value, that would help. I would be able to keep the combination with a lower reach limit.

dougg0k, 2025-03-10:
[image]

Gnarus-G, 2025-03-10:
That doesn't make sense to me. I'm new to Motivity and Synchronous curves, but if the Motivity parameter is defined to be the Max sens like I read in the raw accel discord, the isn't that the limit?