Gnarus-G/maccel #63: CLI improvements

dougg0k, 2025-03-10:
Hi,

Could you make some CLI improvements?

1. In the graph, when there are some high ratio values, the ratio values in between are hard to know what are the value. 
[image]
2. Improve the input highlight with color rather than being bold, as I dont see anything here, only know which are selected by using the `i`.
3. Add description per selected input, in the bottom where you have the keyboard shortcuts, to better understand what each input mean and what it does.
4. Ideally the input would not need the `i`, but rather just be navigated to it with the arrows, with that already be read to be changed or not.

Thanks.

Gnarus-G, 2025-03-10:
1. You already have another thread talking about ratio. How is this different? Also it's a pretty confusing sentence. I guess English is not your first language but I can't help if I don't understand you.

2. and 3. are good requests.

4. I'm definitely not doing this because that would be a terrible design.

dougg0k, 2025-03-10:
1. You complained in that thread, so I added all here with the others.
4. Maybe replace with `enter` instead then.

closed 2025-03-13 by dougg0k