Gnarus-G/maccel #65: Remove install.sh

pxlsec, 2025-03-12:
I was looking into issue #32 (Automate publishing to AUR). But when browsing the code base i realized that it is very redundant to have both an install.sh script and a Makefile. It also makes it a bit messy to maintain both install.sh and PKGBUILD at the same time.
My suggestion is to move all installation logic to the Makefile, remove install.sh, and change all the installation and compilation logic in the PKGBUILD to just `make DESTDIR="${pkgdir}" install`.
The one liner to install the project would change from `curl -fsSL https://www.maccel.org/install.sh | sudo sh` to `git clone https://github.com/Gnarus-G/maccel.git && cd maccel && make install`

Gnarus-G, 2025-03-12:
As you can see there's more to the install script than the Makefile. I want the Makefile to just sort rawdog everything with as few things in the way, and the users get the one install command that does a bunch of extra shit, i.e dkms.

I don't really see how you would unify those cleanly. If I see PR with enough to convert your idea here I could make a better decision or consideration.

Gnarus-G, 2025-03-12:
Also I wanted to look into dkms tar balls as an angle for distribution to see if we similarly simplify things.