Gnarus-G/maccel #66: Regression: TUI crashes partially when perms are insufficient

sy3c4ll, 2025-03-12:
IIRC it used to display a warning about insufficient perms on the statusline whlie disallowing editing.

Now it does this.

[image]

Now this indeed _is_ user error, and the result of the end user doing something they were explicitly told not to do, but I take problem with the fact that panics are not Rusty. /j

I tried piping stderr to a logfile to see what exactly it was complaining about, and came back with this:

```
thread '<unnamed>' panicked at crates/core/src/inputspeed.rs:20:54:
failed to open /dev/maccel: Os { code: 13, kind: PermissionDenied, message: "Permission denied" }
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
```

Is this indeed a regression? I could have a look if you want.

Gnarus-G, 2025-03-12:
That is definitely a regression. Good catch, thank you.

sy3c4ll, 2025-03-12:
Huh, that was... interesting.

So the problem was this:

[image]

I was under the assumption that writing to _parameters_ was failing, but line 20 of `inputspeed.rs` ended up pointing to `/dev/maccel` instead. It's owned by root and it lacks read permissions, therefore the TUI was crashing because it was unable to read the current input speed to display.

I believe to fix this, we could simply add read permissions to `/dev/maccel` for everyone, maybe set it to 444 because do we even write to `/dev/maccel`? It would be greatly appreciated if you have any whereabouts on setting perms on char drivers.

Or, you know, we could just `sudo chmod 444 /dev/maccel` on `install.sh`?

Gnarus-G, 2025-03-12:
Yeah we totally could just that. Still, this has been working fine this whole time why? 

Need to investigate a fresh install with install.sh and ok build to see what state it leaves the /dev/maccel perms.

I would also verify what perms and group it's created with in the kernel module.

Gnarus-G, 2025-03-12:
Don't sweat this one. I'll handle it and explain the flow when I take another look. 

For now I'd say reinstall with the script and add your self to the Marcel group then logout and back in.

sy3c4ll, 2025-03-12:
Thanks for all the help!

Gnarus-G, 2025-03-13:
The character device driver is created in the kernel module. `/dev/maccel` is a representation of it in the filesystem. Once you've installed the kernel module the file is created.

Gnarus-G, 2025-03-13:
At this point these command declared in `maccel.install`:
```sh
udevadm control --reload-rules
udevadm trigger --subsystem-match=usb --subsystem-match=input --subsystem-match=hid --attr-match=bInterfaceClass=03 --attr-match=bInterfaceSubClass=01 --attr-match=bInterfaceProtocol=02
```
Isn't actually run by `makepkg`.

So the right flow really is to run makepkg and then `usermod -aG $USER` and then reboot (or replug your mouse) to trigger udev rules naturally.

closed 2025-03-13 by Gnarus-G