Gnarus-G/maccel #67: Arch installation method fails with linker error

sy3c4ll, 2025-03-12:
Maybe I'm just stupid, but I can't get the `PKGBUILD` to work.

On `cd`-ing into `maccel` and running `makepkg -si`, it compiles nicely until it gets to `maccel-cli`, where it complains it can't find the symbols in `libmaccel.c` and `libmaccel.rs`, like `fpt_from_float` and `sensitivity_rs`.

The full error log is a bit long:

```
   Compiling anstyle-query v1.1.2
   Compiling anstyle-parse v0.2.6
   Compiling scopeguard v1.2.0
   Compiling once_cell v1.20.3
   Compiling colorchoice v1.0.3
   Compiling is_terminal_polyfill v1.70.1
   Compiling rustix v0.38.44
   Compiling either v1.15.0
   Compiling allocator-api2 v0.2.21
   Compiling anstream v0.6.18
   Compiling shlex v1.3.0
   Compiling equivalent v1.0.2
   Compiling clap_lex v0.7.4
   Compiling linux-raw-sys v0.4.15
   Compiling foldhash v0.1.4
   Compiling bitflags v2.9.0
   Compiling clap_builder v4.5.31
   Compiling cc v1.2.16
   Compiling itertools v0.13.0
   Compiling hashbrown v0.15.2
   Compiling tracing-core v0.1.33
   Compiling itoa v1.0.15
   Compiling unicode-segmentation v1.12.0
   Compiling anyhow v1.0.97
   Compiling ryu v1.0.20
   Compiling indoc v2.0.6
   Compiling quote v1.0.39
   Compiling signal-hook-registry v1.4.2
   Compiling syn v2.0.99
   Compiling lock_api v0.4.12
   Compiling mio v1.0.3
   Compiling signal-hook-mio v0.2.4
   Compiling parking_lot v0.12.3
   Compiling static_assertions v1.1.0
   Compiling unicode-width v0.1.14
   Compiling lru v0.12.5
   Compiling unicode-width v0.2.0
   Compiling crossterm v0.28.1
   Compiling cassowary v0.3.0
   Compiling lazy_static v1.5.0
   Compiling overload v0.1.1
   Compiling pin-project-lite v0.2.16
   Compiling unicode-truncate v1.1.0
   Compiling sharded-slab v0.1.7
   Compiling nu-ansi-term v0.46.0
   Compiling tracing-log v0.2.0
   Compiling thread_local v1.1.8
   Compiling castaway v0.2.3
   Compiling compact_str v0.8.1
   Compiling tracing-subscriber v0.3.19
   Compiling maccel-core v0.0.0 (/home/ferris/opt/maccel/src/maccel/crates/core)
   Compiling darling_core v0.20.10
   Compiling clap_derive v4.5.28
   Compiling strum_macros v0.26.4
   Compiling tracing-attributes v0.1.28
   Compiling darling_macro v0.20.10
   Compiling darling v0.20.10
   Compiling instability v0.3.7
   Compiling strum v0.26.3
   Compiling tracing v0.1.41
   Compiling clap v4.5.31
   Compiling clap_complete v4.5.46
   Compiling ratatui v0.29.0
   Compiling tui-input v0.11.1
   Compiling maccel-tui v0.0.0 (/home/ferris/opt/maccel/src/maccel/tui)
   Compiling maccel-cli v0.5.2 (/home/ferris/opt/maccel/src/maccel/cli)
error: linking with `cc` failed: exit status: 1
  |
  = note: LC_ALL="C" PATH="/home/ferris/.rustup/toolchains/stable-x86_64-unknown-linux-gnu/lib/rustlib/x86_64-unk
  = note: some arguments are omitted. use `--verbose` to show all linker arguments
  = note: /usr/bin/ld: /home/ferris/opt/maccel/src/target/release-with-debug/deps/libmaccel_core-4504249f7c684283
          /home/ferris/opt/maccel/src/maccel/crates/core/src/libmaccel.rs:23:(.text._ZN11maccel_core8sens_fns11se
          /usr/bin/ld: /home/ferris/opt/maccel/src/target/release-with-debug/deps/libmaccel_core-4504249f7c684283
          /home/ferris/opt/maccel/src/maccel/crates/core/src/sens_fns.rs:44:(.text._ZN11maccel_core8sens_fns11sen
          /usr/bin/ld: /home/ferris/opt/maccel/src/target/release-with-debug/deps/libmaccel_core-4504249f7c684283
          /home/ferris/opt/maccel/src/maccel/crates/core/src/libmaccel.rs:17:(.text._ZN11maccel_core8sens_fns11se
          /usr/bin/ld: /home/ferris/opt/maccel/src/target/release-with-debug/deps/libmaccel_core-4504249f7c684283
          /home/ferris/opt/maccel/src/maccel/crates/core/src/libmaccel.rs:23:(.text._ZN101_$LT$maccel_core..param
          /usr/bin/ld: /home/ferris/opt/maccel/src/maccel/crates/core/src/libmaccel.rs:23:(.text._ZN11maccel_core
          /usr/bin/ld: /home/ferris/opt/maccel/src/maccel/crates/core/src/libmaccel.rs:23:(.text._ZN11maccel_core
          /usr/bin/ld: /home/ferris/opt/maccel/src/maccel/crates/core/src/libmaccel.rs:23:(.text._ZN11maccel_core
          /usr/bin/ld: /home/ferris/opt/maccel/src/maccel/crates/core/src/libmaccel.rs:23:(.text._ZN11maccel_core
          /usr/bin/ld: /home/ferris/opt/maccel/src/target/release-with-debug/deps/libmaccel_core-4504249f7c684283
          /usr/bin/ld: /home/ferris/opt/maccel/src/target/release-with-debug/deps/libmaccel_core-4504249f7c684283
          /home/ferris/opt/maccel/src/maccel/crates/core/src/inputspeed.rs:21:(.text._ZN11maccel_core10inputspeed
          /usr/bin/ld: /home/ferris/opt/maccel/src/target/release-with-debug/deps/libmaccel_core-4504249f7c684283
          /home/ferris/opt/maccel/src/maccel/crates/core/src/libmaccel.rs:17:(.text._ZN11maccel_core9libmaccel8fi
          /usr/bin/ld: /home/ferris/opt/maccel/src/target/release-with-debug/deps/libmaccel_core-4504249f7c684283
          /home/ferris/opt/maccel/src/maccel/crates/core/src/libmaccel.rs:23:(.text._ZN88_$LT$maccel_core..libmac
          /usr/bin/ld: /home/ferris/opt/maccel/src/target/release-with-debug/deps/libmaccel_core-4504249f7c684283
          /home/ferris/opt/maccel/src/maccel/crates/core/src/libmaccel.rs:50:(.text._ZN11maccel_core9libmaccel8fir$GT$8try_from17h6428034b6322d0b5E+0xf): undefined reference to `fpt_to_str'
          /usr/bin/ld: /home/ferris/opt/maccel/src/target/release-with-debug/deps/libmaccel_core-4504249f7c684283.rlib(maccel_core-4504249f7c684283.maccel_core.86ca93b0f7aebfe-cgu.13.rcgu.o): in function `<maccel_core::libmaccel::fixedptc::Fpt as core::str::traits::FromStr>::from_str':
          /home/ferris/opt/maccel/src/maccel/crates/core/src/libmaccel.rs:62:(.text._ZN84_$LT$maccel_core..libmaccel..fixedptc..Fpt$u20$as$u20$core..str..traits..FromStr$GT$8from_str17h60f175a83b5ec4dcE+0x42): undefined reference to `str_to_fpt'
          /usr/bin/ld: /home/ferris/opt/maccel/src/target/release-with-debug/deps/libmaccel_core-4504249f7c684283.rlib(maccel_core-4504249f7c684283.maccel_core.86ca93b0f7aebfe-cgu.13.rcgu.o): in function `maccel_core::libmaccel::fixedptc::<impl core::convert::From<maccel_core::libmaccel::fixedptc::Fpt> for f64>::from':
          /home/ferris/opt/maccel/src/maccel/crates/core/src/libmaccel.rs:17:(.text._ZN11maccel_core6params7persist86_$LT$impl$u20$core..fmt..Display$u20$for$u20$maccel_core..libmaccel..fixedptc..Fpt$GT$3fmt17h1cd4fbcc0fdef563E+0xf): undefined reference to `fpt_to_float'
          collect2: error: ld returned 1 exit status

  = note: some `extern` functions couldn't be found; some native libraries may need to be installed or have their path specified
  = note: use the `-l` flag to specify native libraries to link
  = note: use the `cargo:rustc-link-lib` directive to specify the native libraries to link with Cargo (see https://doc.rust-lang.org/cargo/reference/build-scripts.html#rustc-link-lib)

error: could not compile `maccel-cli` (bin "maccel") due to 1 previous error
```

Gnarus-G, 2025-03-12:
Another catch. Most likely the build-rs in core ends up with the wrong relative in the make kg run. Probably an easy fix.

sy3c4ll, 2025-03-12:
I've had a look myself, unfortunately I'm not that familiar with `build.rs` and my only guess failed. Hope somebody else comes along and picks this up.

pxlsec, 2025-03-12:
Found [this](https://bbs.archlinux.org/viewtopic.php?id=294150). Add !lto in the options array in the pkgbuild, I had the build error on my machine as well and this seemed to fix it.

closed 2025-03-13 by Gnarus-G