Gnarus-G/maccel #68: Support scrolling on parameter input widget

sy3c4ll, 2025-03-12:
I don't know ratatui and I don't know how big a change this would get, so I'm hesitant in suggesting this.

On my peasant 13-inch laptop screen, this is all that shows for the Synchronous mode:

[image]

I happened to experience a similiar problem when making the GUI prototype, which I solved by making the column scrollable. Now I know scrolling with a mouse in a TUI is weird, but I imagine something like automatically scrolling the view down on pressing Tab could work.

If this is unexpectedly difficult a change, or if you have a better idea in mind, please do educate me.

Gnarus-G, 2025-03-12:
Scrolling might be the only way. It's not great though but what can you do. I personally just zoom out when I start the TUI.

I still want to first come up with a way to be remain kbd first.

dougg0k, 2025-03-13:
https://ratatui.rs/examples/widgets/scrollbar/ with Keycode?