Gnarus-G/maccel #78: Add gain version of `Synchronous`

redactedontop, 2025-05-18:
Howdy!

I really need a gain variant of Synchronous for what I do. Could that be possible?

Thanks,
Alex <3

Gnarus-G, 2025-05-19:
It could be, but I'm curious why you need it.

redactedontop, 2025-05-21:
Because it's basically gain + Motivity (when smooth and gamma are 1), which is my prefered acceleration curve. Just need a gain toggle in Synchronous.

redactedontop, 2025-05-21:
I used it for FPS games all the time back in Windows, but non-gain doesn't feel the same...

umbrageodotus, 2025-05-24:
I can help, I know rust very well. What's your discord?

Gnarus-G, 2025-05-24:
You'll spend more time writing c code than rust code though. https://discord.gg/GzWVtKtC

umbrageodotus, 2025-06-28:
Howdy. Has this been implemented? I can't help atm and I need this because libinput SUCKS.