Gnarus-G/maccel #80: Unable to install maccel on SteamOS 3.6

Duartefon, 2025-07-30:
I'm trying to install maccel on the Steam Deck Oled. I've already installed all the dependencies, however when I try to run the install.sh script I get the following error:

"Error! Your kernel headers for kernel 6.11.11-valve20-1-neptune-611-gd35c3ed359a0 cannot be found at /usr/lib/modules/6.11.11-valve20-1-neptune-611-gd35c3ed359a0/build or /usr/lib/modules/6.11.11-valve20-1-neptune-611-gd35c3ed359a0/source.
Please install the linux-headers-6.11.11-valve20-1-neptune-611-gd35c3ed359a0 package or use the --kernelsourcedir option to tell DKMS where it's located."
 
Even after installing  the linux headers I still got the same error.

Duartefon, 2025-07-31:
Thanks! I already use maccel on my desktop and love it! However was not sucessful on installing on the steam deck :P How would I use the --kernelsourcedir flag?

Duartefon, 2025-07-31:
I ran the command "curl -fsSL https://www.maccel.org/install.sh | sudo sh" to install the driver. The script failed on the "Installing the driver (kernel module)" step. 
The entire error message I got was:

"Module maccel 0.5.4 is not installed for kernel 6.11.11-valve20-1-neptune-611-gd35c3ed359a0 (x86_64). Skipping...
Module maccel 0.5.4 is not built for kernel 6.11.11-valve20-1-neptune-611-gd35c3ed359a0 (x86_64). Skipping...
Deleting module maccel-0.5.4 completely from the DKMS tree.
Sign command: /usr/lib/modules/6.11.11-valve20-1-neptune-611-gd35c3ed359a0/build/scripts/sign-file
Binary /usr/lib/modules/6.11.11-valve20-1-neptune-611-gd35c3ed359a0/build/scripts/sign-file not found, modules won't be signed
Creating symlink /var/lib/dkms/maccel/0.5.4/source -> /usr/src/maccel-0.5.4
Error! Your kernel headers for kernel 6.11.11-valve20-1-neptune-611-gd35c3ed359a0 cannot be found at /usr/lib/modules/6.11.11-valve20-1-neptune-611-gd35c3ed359a0/build or /usr/lib/modules/6.11.11-valve20-1-neptune-611-gd35c3ed359a0/source.
Please install the linux-headers-6.11.11-valve20-1-neptune-611-gd35c3ed359a0 package or use the --kernelsourcedir option to tell DKMS where it's located."

Then I tried to download different linux header from valve SteamOS repo and put it on /usr/lib/modules/ where I have the following installs:

drwxr-xr-x 1 root root     10 Jul 23 12:46 5.13.0-valve37-1-neptune
drwxr-xr-x 1 root root     10 Jul 23 14:59 6.11.11-valve19-1-neptune-611-g88b36d49a5e3
drwxr-xr-x 1 root root    500 Jul 23 15:12 6.11.11-valve20-1-neptune-611-gd35c3ed359a0
drwxr-xr-x 1 root root     10 Jul 23 12:44 6.12.8-arch1-1
drwxr-xr-x 1 root root     10 Jul 23 12:31 6.1.52-valve19-1-neptune-61
drwxr-xr-x 1 root root     10 Jul 23 14:49 6.5.0-valve23-1-neptune-65-gd4bfbf9dfafe

My current kernel is 6.11.11-valve20-1-neptune-611-gd35c3ed359a0

Gnarus-G, 2025-07-31:
What command did you use to install?
Also did you use `uname -r` to see your current kernel?

Duartefon, 2025-07-31:
yes, `uname -r` returns `6.11.11-valve20-1-neptune-611-gd35c3ed359a0`.

If by install you mean how I installed the linux header, after downloading it I ran `sudo pacman -U /path/to/package.pkg.tar.zst`

Duartefon, 2025-08-01:
I finally managed to install it! basically what was happening is I had a kernel version that no matter what linux header provided by Valve I installed, was never the correct version for that kernel. The solution I found was to install the kernel for that linux header version, instead of the other way around. I'm now gonna test it to see if everything is working smoothly!

closed 2025-08-04 by Gnarus-G