Gnarus-G/maccel #87: Maccel kernel panic

Isser-yts, 2025-08-26:
Panic Report
Arch: x86_64
Version: 6.16.3-2-cachyos

[  366.589353] CPU: 15 UID: 0 PID: 0 Comm: swapper/15 Tainted: G           OE       6.16.3-2-cachyos #1 PREEMPT(full)  3921225e34afae8f8913219140551f3afefd9131
[  366.589357] Tainted: [O]=OOT_MODULE, [E]=UNSIGNED_MODULE
[  366.589358] Hardware name: LENOVO 83LT/LNVNB161216, BIOS RLCN20WW 04/14/2025
[  366.589360] RIP: 0010:maccel_events+0x3dc/0x21e0 [maccel]
[  366.589366] Code: 44 39 d6 7c ea 45 85 c9 0f 85 3f 02 00 00 45 31 e4 48 83 f9 ff 74 20 48 89 c8 48 8b 5c 24 58 48 c1 f9 20 48 c1 e0 20 48 89 ca <48> f7 fb 48 89 05 1a cc 5b 00 49 89 c4 48 8d bc 24 f0 00 00 00 48
[  366.589367] RSP: 0018:ffffce49c05e0be0 EFLAGS: 00010817
[  366.589370] RAX: 6a09e66700000000 RBX: 0000000000000000 RCX: 0000000000000001
[  366.589371] RDX: 0000000000000001 RSI: 0000000000000011 RDI: 0000000000000000
[  366.589372] RBP: ffffce49c05e0d48 R08: 0000000000000002 R09: 0000000000000000
[  366.589374] R10: 0000000000000011 R11: 0000000000000000 R12: 0000000000000000
[  366.589375] R13: 0000000100000000 R14: ffff8c9b9a48c30c R15: ffff8c9b9a48c310
[  366.589376] FS:  0000000000000000(0000) GS:ffff8cab0ba89000(0000) knlGS:0000000000000000
[  366.589378] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[  366.589379] CR2: 0000056db6c00000 CR3: 0000000da3e24000 CR4: 0000000000f50ef0
[  366.589381] PKRU: 55555554
[  366.589382] Call Trace:
[  366.589384]  <IRQ>
[  366.589389]  ? srso_alias_return_thunk+0x5/0xfbef5
[  366.589393]  ? tcp_v4_do_rcv+0x164/0x2c0
[  366.589397]  input_pass_values+0x141/0x160
[  366.589402]  ? __pfx_input_repeat_key+0x10/0x10
[  366.589404]  input_handle_event+0x192/0x1a0
[  366.589407]  input_repeat_key+0x167/0x1a0
[  366.589409]  ? __pfx_input_repeat_key+0x10/0x10
[  366.589410]  call_timer_fn+0x26/0x120
[  366.589415]  timer_expire_remote+0x40e/0x540
[  366.589419]  tmigr_handle_remote+0x3a8/0x450
[  366.589424]  handle_softirqs+0xe6/0x2a0
[  366.589428]  irq_exit_rcu+0xd8/0x100
[  366.589430]  sysvec_apic_timer_interrupt+0x71/0x90
[  366.589434]  </IRQ>
[  366.589435]  <TASK>
[  366.589436]  asm_sysvec_apic_timer_interrupt+0x1a/0x20
[  366.589438] RIP: 0010:cpuidle_enter_state+0xc2/0x7f0
[  366.589440] Code: 00 00 e8 b1 49 e4 fe e8 dc f2 ff ff 49 89 c4 0f 1f 44 00 00 31 ff e8 4d 63 e2 fe 45 84 ff 0f 85 d2 04 00 00 fb 0f 1f 44 00 00 <45> 85 f6 0f 88 89 02 00 00 49 63 f6 4c 89 e2 48 2b 14 24 48 6b ce
[  366.589442] RSP: 0018:ffffce49c024fe78 EFLAGS: 00000246
[  366.589443] RAX: ffff8cab0ba89000 RBX: 0000000000000002 RCX: 000000555a6d3a6b
[  366.589445] RDX: 000000555a6d3a6b RSI: 0000000000000014 RDI: 0000000000000000
[  366.589446] RBP: ffff8c9b8755b000 R08: ffff8caabd9f3240 R09: 000000555a6d3a6b
[  366.589447] R10: ffffffffffffffff R11: 00000000000003cb R12: 000000555a6d3a6b
[  366.589448] R13: ffffffffb13fcbe0 R14: 0000000000000002 R15: 0000000000000000
[  366.589451]  ? cpuidle_enter_state+0xb3/0x7f0
[  366.589453]  cpuidle_enter+0x31/0x50
[  366.589457]  do_idle+0x1cd/0x240
[  366.589461]  cpu_startup_entry+0x29/0x30
[  366.589463]  start_secondary+0x119/0x140
[  366.589466]  common_startup_64+0x13e/0x141
[  366.589472]  </TASK>
[  366.589473] Modules linked in: nft_queue nft_ct nfnetlink_queue snd_seq_dummy snd_hrtimer snd_seq snd_seq_device nft_chain_nat xt_mark xt_cgroup xt_MASQUERADE nf_nat nf_conntrack nf_defrag_ipv6 nf_defrag_ipv4 ip6t_REJECT nf_reject_ipv6 ipt_REJECT nf_reject_ipv4 xt_tcpudp nft_compat nf_tables cmac algif_hash algif_skcipher af_alg bnep snd_hda_codec_realtek snd_hda_codec_generic snd_hda_scodec_component snd_hda_codec_hdmi vfat fat snd_sof_amd_acp70 snd_sof_amd_acp63 snd_sof_amd_vangogh snd_sof_amd_rembrandt snd_sof_amd_renoir snd_sof_amd_acp snd_sof_pci snd_sof_xtensa_dsp snd_sof snd_sof_utils snd_pci_ps snd_soc_acpi_amd_match snd_amd_sdw_acpi soundwire_amd soundwire_generic_allocation amd_atl intel_rapl_msr mt7925e soundwire_bus intel_rapl_common mt7925_common snd_soc_sdca mt792x_lib snd_soc_core mt76_connac_lib snd_compress mt76 snd_hda_intel ac97_bus snd_pcm_dmaengine snd_intel_dspcfg snd_intel_sdw_acpi snd_rpl_pci_acp6x uvcvideomac80211 btusb snd_acp_pci snd_hda_codec videobuf2_vmalloc btrtl snd_amd_acpi_mach uvc
[  366.589523]  libarc4 snd_hda_core videobuf2_memops btintel snd_acp_legacy_common kvm_amd videobuf2_v4l2 snd_hwdep snd_pci_acp6x btbcm r8169 btmtk videobuf2_common snd_pcm realtek ucsi_acpi cfg80211 ideapad_laptop spd5118 kvm bluetooth typec_ucsi snd_timer snd_pci_acp5x mdio_devres videodev typec platform_profile snd_rn_pci_acp3x libphy snd hid_multitouch i2c_piix4 snd_acp_config roles sparse_keymap maccel(OE) joydev mousedev mc irqbypass snd_soc_acpi thunderbolt i2c_hid_acpi psmouse rapl nvidia_wmi_ec_backlight soundcore k10temp mdio_bus lenovo_wmi_hotkey_utilities snd_pci_acp3x pcspkr rfkill wmi_bmof i2c_smbus mac_hid wdat_wdt i2c_hid pkcs8_key_parser ntsync i2c_dev sg crypto_user loop nfnetlink lz4 zram 842_decompress 842_compress lz4hc_compress lz4_compress ip_tables x_tables dm_crypt encrypted_keys trusted asn1_encoder tee dm_mod amdgpu amdxcp i2c_algo_bit drm_exec polyval_clmulni gpu_sched ghash_clmulni_intel drm_suballoc_helper sha512_ssse3 drm_panel_backlight_quirks sha1_ssse3 nvme drm_buddy aesni_intel
[  366.589585]  nvme_core serio_raw drm_display_helper nvme_keyring ccp cec nvme_auth nvidia_drm(O) drm_ttm_helper ttm nvidia_uvm(O) nvidia_modeset(O) video wmi nvidia(O)
[  366.589598] ---[ end trace 0000000000000000 ]---
[  366.589599] RIP: 0010:maccel_events+0x3dc/0x21e0 [maccel]
[  366.589602] Code: 44 39 d6 7c ea 45 85 c9 0f 85 3f 02 00 00 45 31 e4 48 83 f9 ff 74 20 48 89 c8 48 8b 5c 24 58 48 c1 f9 20 48 c1 e0 20 48 89 ca <48> f7 fb 48 89 05 1a cc 5b 00 49 89 c4 48 8d bc 24 f0 00 00 00 48
[  366.589603] RSP: 0018:ffffce49c05e0be0 EFLAGS: 00010817
[  366.589604] RAX: 6a09e66700000000 RBX: 0000000000000000 RCX: 0000000000000001
[  366.589605] RDX: 0000000000000001 RSI: 0000000000000011 RDI: 0000000000000000
[  366.589607] RBP: ffffce49c05e0d48 R08: 0000000000000002 R09: 0000000000000000
[  366.589608] R10: 0000000000000011 R11: 0000000000000000 R12: 0000000000000000
[  366.589609] R13: 0000000100000000 R14: ffff8c9b9a48c30c R15: ffff8c9b9a48c310
[  366.589610] FS:  0000000000000000(0000) GS:ffff8cab0ba89000(0000) knlGS:0000000000000000
[  366.589611] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[  366.589612] CR2: 0000056db6c00000 CR3: 0000000da3e24000 CR4: 0000000000f50ef0
[  366.589613] PKRU: 55555554
[  366.589614] Kernel panic - not syncing: Fatal exception in interrupt
[  366.590079] Kernel Offset: 0x2d800000 from 0xffffffff81000000 (relocation range: 0xffffffff80000000-0xffffffffbfffffff)

Isser-yts, 2025-08-28:
Okay, warp terminal gave me an opinion that this kernel panic because of division in maccel driver like 00000/06w5b242 kind of thing. Or that I need to disable touchpad to not encounter this kernel panic. I didn't disabled touchpad so I've tried leetmouse It makes mouse feel very precise and quake-like. Really like it. It works without issues.

dougg0k, 2025-08-28:
Imo no other option works as good as maccel on linux.

Isser-yts, 2025-08-28:
Did you try it? Try [leetmouse](https://github.com/systemofapwne/leetmouse) once, and then justify what you think "IMO".

pxlsec, 2025-09-18:
Hey, I took some time to look into the issue.
After some disassembly I think the issue stems from the function `input_speed(fpt dx, fpt dy, fpt time_ms)`, it has [a line](https://github.com/Gnarus-G/maccel/blob/43b624c72b04a62f55155b99c2ebe2f531abc1b9/driver/speed.h#L25C1-L25C42) where it divides the magnitude of the input with time_ms without checking whether time_ms is zero. 

I also did some digging to find out why time_ms was zero in the first place and stumbled upon this:
>ktime_t ktime_get(void)[¶](https://www.kernel.org/doc/html/latest/core-api/timekeeping.html#c.ktime_get)
>
>    CLOCK_MONOTONIC
>
>    Useful for reliable timestamps and measuring short time intervals accurately. Starts at system boot time but stops during suspend.

The fix would be to check time_ms before dividing.